KoynovStas/onvif_srvd #1: I tried to use this server,find  has no login ,can use this server. why??

yyong-brs, 2017-04-13:


KoynovStas, 2017-04-13:
The server has only the minimum implementation and validation there is no password.

If you need support login + password, you must compile the server with support for WS-security:

```
make clean
make WSSE_ON=1
```

if you need it you can implement it something like this:

```C
// check authentification
const char *username = soap_wsse_get_Username(this->soap);
if (!username)
	return this->soap->error;
if (ctx->user != username)
	return SOAP_FAULT;
if (soap_wsse_verify_Password(this->soap, ctx->password.c_str()))
	return this->soap->error;
```

this validation code can be put into func request video sources:

`int MediaBindingService::GetVideoSources`

yyong-brs, 2017-04-13:
ok,thanks!

yyong-brs, 2017-04-13:
But,now  i found that  ptz can not work, if i need import the ptz wsdl or something else ???

KoynovStas, 2017-04-13:
Yes, you have to download from the website: [https://www.onvif.org/profiles/specifications/](https://www.onvif.org/profiles/specifications/) the wsdl file for the PTZ. And put in the directory wsdl.

The next step is to add the implementation file for this service. For example `$(COMMON_DIR)/ServicePTZ.cpp` and add this file to the list of `SOURCES` in the Makefile.

yyong-brs, 2017-04-13:
Thank you, I am also just beginning to learn this, later also asked you to help!!!!

KoynovStas, 2017-04-24:
The issue was resolved? If yes, then I close it.

closed 2017-05-01 by KoynovStas

thewh1teagle, 2020-11-09:
Hi
I tried to compile it with 
```
// check authentification
const char *username = soap_wsse_get_Username(this->soap);
if (!username)
	return this->soap->error;
if (ctx->user != username)
	return SOAP_FAULT;
if (soap_wsse_verify_Password(this->soap, ctx->password.c_str()))
	return this->soap->error;
```
I added this to that function:
```
int MediaBindingService::GetVideoSources
```

And i got this erros:
```
src/ServiceMedia.cpp: In member function 'virtual int MediaBindingService::GetVideoSources(_trt__GetVideoSources*, _trt__GetVideoSourcesResponse&)':
src/ServiceMedia.cpp:44:28: error: 'soap_wsse_get_Username' was not declared in this scope
     const char *username = soap_wsse_get_Username(this->soap);
                            ^~~~~~~~~~~~~~~~~~~~~~
src/ServiceMedia.cpp:44:28: note: suggested alternative: 'soap_get__QName'
     const char *username = soap_wsse_get_Username(this->soap);
                            ^~~~~~~~~~~~~~~~~~~~~~
                            soap_get__QName
src/ServiceMedia.cpp:49:9: error: 'soap_wsse_verify_Password' was not declared in this scope
     if (soap_wsse_verify_Password(this->soap, ctx->password.c_str()))
         ^~~~~~~~~~~~~~~~~~~~~~~~~
src/ServiceMedia.cpp:49:9: note: suggested alternative: 'soap_write__wsse__Password'
     if (soap_wsse_verify_Password(this->soap, ctx->password.c_str()))
         ^~~~~~~~~~~~~~~~~~~~~~~~~
         soap_write__wsse__Password
Makefile:155: recipe for target 'src/ServiceMedia.o' failed
make: *** [src/ServiceMedia.o] Error 1
root@ef2cafe9927a:/src#
 ```

KoynovStas, 2020-11-10:
Dude, I don't believe you can't solve this problem!
These are not my functions! These are functions from gsoap. You can read their documentation!
And understand what an include is needed: `#include "wsseapi.h"`

Then you can re-read the readme of this repository and understand that you need to rebuild the gsoap.

I believe in you, you can do it!