KoynovStas/onvif_srvd #2: Error getting file gsoap_2.8.55.zip from SF

ZigFisher, 2018-02-25:
Hi !

There is a problem in getting the file gsoap_2.8.55.zip from the resource sourceforge.net 
Do you know the alternative ? Or can you put this file to yourself ?

TNX !

KoynovStas, 2018-02-25:
This is the sourceforge.net problem. You can write to them in support. 
Here is a link to a copy of the file: [gsoap_2.8.55.zip](https://www.dropbox.com/s/fhgp3uxyxsxrxg1/gsoap_2.8.55.zip?dl=0)
You will need to save this file in the SDK/gsoap.zip directory of the project.

To save large binary files in the repository is a bad tone. But if this problem happens again, I will have to think of a way, possibly breaking the rule.

ZigFisher, 2018-02-25:
TNX !

closed 2018-02-25 by ZigFisher