KoynovStas/onvif_srvd #3: Compilation Issue on Raspberry Pi

HanYangZhao, 2018-03-27:
Hi,

I'm trying to create a RPi based Onvif camera, but ran into some issues during compilation. Basically the compilations get stuck at :
`  [build]  generated/soapC.o:
g++ -c generated/soapC.cpp -o generated/soapC.o  -s  -DDAEMON_NAME='"onvif_srvd"' -DDAEMON_MAJOR_VERSION=1 -DDAEMON_MINOR_VERSION=1 -DDAEMON_PATCH_VERSION=0 -DDAEMON_PID_FILE_NAME='"onvif_srvd.pid"' -DDAEMON_LOG_FILE_NAME='"onvif_srvd.log"' -DDAEMON_NO_CHDIR=1 -DDAEMON_NO_CLOSE_STDIO=0 -I./src -I./generated -I./gsoap-2.8/gsoap -I./gsoap-2.8/gsoap/custom -I./gsoap-2.8/gsoap/plugin -I./gsoap-2.8/gsoap/import -std=c++11 -O2  -Wall  -pipe
` 
I'm suspecting it's because the RPi ran out of ram (only 1GB). Do you have any idea how to resolve this problem?  I have tried playing around with the memory splitting and swap file size, but nothing seems to work. Is the compilation process multi- threaded or single threaded?

Thanks

KoynovStas, 2018-03-27:
1. Build will be multithreaded (recursive make) if you run make with the `-j` option, for example:
`make -j8 #make will use 8 processes!`
by default, make works in single-threaded mode.

2. You can remove the `-pipe` option which uses PIPE to transfer temporary files between the build processes. (in theory this should save RAM)

3. Why do you even compile an application on the RPi itself?
I make for embeded systems using a cross compiler on a large Linux. You need to download a cross-compiler for RPi and assemble the application on your PC. and then copy the binary to RPi.
In this case, the `SOAPCPP2` and `WSDL2H` applications will be compiled for your large Linux PC. but the build of the binary will also be done on Linux PC, but for RPi. In this case, you will not have problems with the size of RAM.

HanYangZhao, 2018-03-27:
Thanks for the quick reply.  I wanted to compile on the Pi because I don't have a cross compile workflow setup right now. But since removing the -pipe didn't do anything, guess I'll have to cross compile

jemyzhang, 2019-02-11:
The compilation of soapC needs up to 5 Gb memory as I saw on the host machine...

Micah2015, 2019-05-09:
I compile it on Pi successfully when I change the swap to 2GB, it takes me over 1 hour to finish it. You'd better to compile it on PC, `make GCC=arm-linux-gnueabihf-g++`

parinpatel, 2020-02-29:
@HanYangZhao 
Hi, you can also remove unnecessary GSOAP lib files which are not required for compilation, Please remove and rebuild your application with this optimization, it will resolve your issue.

moykky, 2020-09-21:
> @HanYangZhao
> Hi, you can also remove unnecessary GSOAP lib files which are not required for compilation, Please remove and rebuild your application with this optimization, it will resolve your issue.

Would this help me with binary file size? I'm cross compiling this for my ipcam and compiled binary size is 6.1mb, it's way too much for 8mb SPI flash...

KoynovStas, 2020-09-22:
> Would this help me with binary file size? I'm cross compiling this for my ipcam and compiled binary size is 6,4 MB, it's way too much for 8 MB SPI flash...

No. This is a gsoap problem. It generates a parser which is very bold. If you need ONVIF on a small device. You have to write the server/client and parser yourself without using the gsoap. But this is a long and difficult journey!

moykky, 2020-09-22:
> No. This is a gsoap problem. It generates a parser which is very bold. If you need ONVIF on a small device. You have to write the server/client and parser yourself without using the gsoap. But this [...]

Thank you.

parinpatel, 2020-09-22:
Please use strip command for further reduction of binary size

KoynovStas, 2020-09-23:
> @moykky
> Please use strip command for further reduction of binary size

6-8MiB is already the size after the strip. The binary with debug information reaches > 20MiB. I say gsoap generates a Very Bold binary. and is only suitable for Large systems. Or as a prototype for a "quick" test

parinpatel, 2020-09-23:
@KoynovStas ok