KoynovStas/onvif_srvd #4: Extra stream in webservice

cassiomaes, 2018-08-14:
Hi, how i add a extra stream in the service? ex: two urls http://ipcamera:554/onvif1  and http://ipcamera:554/onvif2 in the same port service.?

this two streams is from the same camera with 2 diferent resolutions.  

thanks.

KoynovStas, 2018-08-15:
You must create two profiles (Profile Media Services). This daemon does this according to (`./onvif_srvd --help`)

by specifying the arguments cmd:
`--name RTSP1 --width 800 --height 600 --url rtsp: //% s: 554 / unicast_1 --type JPEG`
`--name RTSP2 --width 640 --height 480 --url rtsp: //% s: 554 / unicast_2 --type JPEG`

The beginning of the profile is the argument `--name`, end of the profile is the argument `--type`. 
When the `--type` argument is encountered, a media profile will be created, according to the early parameters and added to the ONVIF (it will be added to list of Profile Media Services).

**full example:**
```
./onvif_srvd  --ifs eth0 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP1 --width 800 --height 600 --url rtsp://%s:554/unicast_1 --type JPEG  --name RTSP2 --width 640 --height 480 --url rtsp://%s:554/unicast_2 --type JPEG
```

**Note:**

> Name of profiles **must be unique!**

PS. Do not be afraid to learn and read the source.

cassiomaes, 2018-08-16:
hi...  I thought the profile name was reserved... my mistake. i change the
cmdline call with other profile name and works.

But the profiles were created separately. My real question is if the two
profiles can stay togheter in a 'group'.

Let me explain...

I use the program Onvif Device Manager 2.2.250 to check a device with onvif
native support and two profiles are in the same 'group'  (perhaps better
'subprofile')

profiles MainStream and SubStream

With the onvif_srvd the two profiles are created separately fig2 (.profiles
extra and extra2)

it is possivle to pass parameters so that it works like in (fig 1) ?

working in rapsberry pi zero with chinese cameras without onvif.  :)

thanks for the reply!

cassiomaes, 2018-08-16:
[image]
[image]

KoynovStas, 2018-08-17:
Ok I understood what you want. 
Forgive me but I did not need this functionality. And I dont did it :(.

You will need to consult the documentation (specifications) of ONVIF. Perhaps you just need to create a subprofile or a second link to the stream URI. To understand this, take a look at the TCP\IP packet analyzer ([wireshark](https://www.wireshark.org)) which query\response you get from your camera (which has what you want), this will help you understand which function you need to add functionality. Next, you will need to correct the cmd processing code so that the necessary functionality works from the command line.

santhku, 2019-05-21:
Im able to compile and now trying to test this build, can you please let me know the steps 
As i 
I executed the below cmd
./onvif_srvd  --ifs eth0 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP --width 800 --height 600 --url rtsp://%s:554/unicast --type JPEG"

but i was unable to get the IP address assigned,  
I tried testing using the system IP address but it dont work

KoynovStas, 2019-05-21:
What does not work? Rtsp? There is no RTSP server in this daemon. This daemon must give you the URL you specified for the request from the client. RTSP server should work on the specified IP address and port.

According to your cmd:
`./onvif_srvd --ifs eth0 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP --width 800 --height 600 --url rtsp://%s:554/unicast --type JPEG`

On your host, there is an interface named eth0. The daemon will attempt to determine the first IP address. eg:

**eth0: 192.168.10.1**

Your link is converted to `rtsp://192.168.10.1:554/unicast`

santhku, 2019-05-23:
Thanks for the quick reply, 
I need ONVIF that supports the rtsp, does this supports RTSP, If yes u
could you Guide me to test the data stream

KoynovStas, 2019-05-23:
No, **this daemon does not support RTSP**. He should not do that! 

[ONVIF (Open Network Video Interface Forum)](https://en.wikipedia.org/wiki/ONVIF)  is a set of specifications for the **COMMUNICATION** of equipment in an automatic (semi-automatic) mode. So that the Client (DVR) can find and AGREE with the server (IP camera).

This daemon will inform the Client which protocols and data streams (video, audio, etc.) the device supports. After receiving this information, the client may request this data. The data transfer protocols MJPEG, RTSP device in most cases are separate applications.

You can find the RTSP server and run it separately, or include it inside this daemon (**but this is a bad practice, the application will have a lot of responsibilities and will be very large**)

santhku, 2019-05-23:
Thanks for the reply, 

I have to video stream on the network and it should be compliant to ONVIF for NVR application, so how can i start. I have no knowledge on how to start , could help me in understanding the rtsp server and test

KoynovStas, 2019-05-26:
I wrote these two projects:

[wsdd](https://github.com/KoynovStas/wsdd) - is Linux daemon for ONVIF [WS-Discovery](https://en.wikipedia.org/wiki/WS-Discovery) service (server side).
[onvif_srvd](https://github.com/KoynovStas/onvif_srvd) - ONVIF Device(IP camera) Service server (Linux daemon)

[wsdd](https://github.com/KoynovStas/wsdd) is a server that responds to the request (who is here) of the DVR. Thus lets the Registrar know what camera is on the network. I executed it as a separate daemon because our device had two Ethernet interfaces. And I run this daemon twice, with different parameters.

[onvif_srvd](https://github.com/KoynovStas/onvif_srvd) is something that "agrees" with the DVR also using [soap](https://en.wikipedia.org/wiki/SOAP) protocol . Thus, the Registrar understands what format, what resolution, port, transmission Protocol, etc

In General, what is [ONVIF (Open Network Video Interface Forum)](https://en.wikipedia.org/wiki/ONVIF) — an industry international organization that develops standardized protocols for the interaction of various equipment and software included in the security systems (IP-cameras, IP-encoders, video recorders, access controllers, etc.).

But it's the lyrics! All this is necessary to standardize the "communication and other mechanisms" between the devices of the video surveillance system. For example, in the simplest case between the IP camera and DVR.
It's for describe the standard (Protocol) as to ask, what to say and in what format. Here we are waiting for a great surprise and disappointment!
Why?
This organization was infiltrated by people from MicroSoft, who promoted the soap standard. It is such an add-in over XML. And the [WSDL](https://ru.wikipedia.org/wiki/WSDL).

Their idea was such that having a [WSDL](https://ru.wikipedia.org/wiki/WSDL) file, you will be able to generate a parser (server) to work with this device in dynamics. Like all simply! But we must have a parser of WSDL files. We must use soap and not clean XML that prevents the use of simple XML parsers. And since both the WSDL file and the soap parser are very complex (if we consider the General case), then we need very large resources, but these people were far from embedded systems, they thought in terms of cloud computing... 
That is like Skype — a monster that can not even perform its functions. (A feeling that to recruit not engineers, and all sorts of Olympic medalists who have a big brain that prevents them from thinking.)

Therefore, to write all these parsers "on the mind" in a short time is not a real task. I used [gsoap](https://www.genivia.com/dev.html) which can generate soap parser from WSDL file. This parser is the server that handles soap requests. You're just adding logic. The drawback — the application is very large!.

However, these applications only need to communicate, the actual useful work they are not doing. After finding the camera it is requested [RTSP](https://en.wikipedia.org/wiki/Real_Time_Streaming_Protocol) address and all. Then RTSP works on its own and has nothing to do with ONVIF.

You say you already have a video stream. If this stream is in RTSP format, then you only need to pass the address (URL) of this stream as a parameter to onvif_srvd so that it passes it to the DVR! If not, then you need to find and run an RTSP server that will stream your video stream. On the Internet you can find ready-made implementations for various video stream options.

I hope you understand my bad English.

santhku, 2019-05-28:
Regret for delayed response and thank you for explaining in detail, I will try out the methods you have suggested me and let you know my finding.

santhku, 2019-06-10:
I am trying, How to enable get / set method for below 
Device: GetNetworkDefaultGateway
Device: GetNetworkInterfaces
Device: GetHostname
Device: GetDNS
Device: GetNTP
Device: GetNetworkProtocols
Device: GetZeroConfiguration
Device: GetDiscoveryMode
I need help on this

damianeroo, 2021-06-28:
Hi, did anyone add substream and mainstream as one profile ?

KoynovStas, 2021-06-28:
So far this is impossible, because there is a rejection for the uniqueness of the name. There is no time to deal with this project yet. Perhaps, when there is time, I will add this feature.