KoynovStas/onvif_srvd #9: Boardcast

santhku, 2019-06-18:
Hi 
I am trying to setup Onvif server on broadcast, could you help

KoynovStas, 2019-06-18:
[Broadcast](https://en.wikipedia.org/wiki/Broadcasting_(networking)) Not used in Onvif. 
You can do it but it **is not right**! The server itself may respond to broadcasts requests, but do a broadcast response **NO**. 
Onvif uses WS-Discovery (Web Services Dynamic Discovery)  which uses [multicast](https://en.wikipedia.org/wiki/Multicast).

Media server like RTSP can use [multicast](https://en.wikipedia.org/wiki/Multicast).

See [ONVIF-Streaming-Spec](https://www.onvif.org/specs/stream/ONVIF-Streaming-Spec-v210.pdf)

santhku, 2019-06-24:
I am facing new issue on my board 
Iam getting an error, 
on "worker" error : "SOAP-ENV:sender"
How can this error can be resolved

santhku, 2019-06-24:
I am getting above error on the client side when probe is called

KoynovStas, 2019-06-24:
1. If this is a new issue, why is the question asked in this issue?
2. If I answer you that you have an error either in the client or in the request handler, or in response will it help you in the answer? Then why is the context completely out of the question? if not me, then maybe someone else could answer, but nobody understands what request, what answer, what client, what conditions, how to reproduce this problem, nothing .... What do you expect to hear?
3. My versions were fully developed for the [ONVIF standard client](https://sourceforge.net/projects/onvifdm/) at the time of writing, perhaps the client requires new changes, you must agree on the protocol versions and specifications.

santhku, 2019-06-25:
Sorry to post here 
I will post the complete details in a new issue listing..

closed 2019-06-25 by santhku