KoynovStas/onvif_srvd #10: on_worker_error : "SOAP-ENV:Sender"

santhku, 2019-06-25:
Hi, 

When i run  ONVIF_SRVD_DEBUG on My PC it work fine without any error with the onvif-device-tool_16.05-20160518_Xenial_amd64 tool, and the same code is cross compiled and ONVIF_SRVD_DEBUG ported to NXP processor and if i run the ONVIF_SRVD_DEBUG, i am getting below error in the debug prints of onvif-device-tool_16.05-20160518_Xenial_amd64 tool, 

Test HTTP digest...
initDevice: HTTP digest support: 0
device support GetServices (ONVIF v2.x)
unhandled service: http://www.onvif.org/ver10/device/wsdl
address: http://127.0.0.1:4000
media service: http://127.0.0.1:4000
on_worker_error : "SOAP-ENV:Sender"
wsdd: probe_done
on_pushButton_probe_clicked
on_listWidget_Devices_currentRowChanged: -1
WSDDThread::probe
addDevice : "http://192.168.36.75:4000" , manual: true
"[Manual]\nsanthosh\nAddress: 192.168.36.75\nFirmware: 1.1\nLocation: "
on_listWidget_Devices_currentRowChanged: 0
on_listWidget_Devices_currentRowChanged: row 0 => manual 0
initDevice: time skew 171311744 sec
now = Mon Jun 24 21:14:16 2019

Test HTTP digest...
initDevice: HTTP digest support: 0
device support GetServices (ONVIF v2.x)
unhandled service: http://www.onvif.org/ver10/device/wsdl
address: http://127.0.0.1:4000
media service: http://127.0.0.1:4000
on_worker_error : "SOAP-ENV:Sender"
wsdd: probe_done

How to resolve this

santhku, 2019-06-25:
I am able to locate the issue but i am unable to resolve it
"  tds__GetServicesResponse.Service.back()->XAddr      = XAddr;" 
The xaddr is not able to send the ipaddress and port 

Below is the full function. 
int DeviceBindingService::GetServices(_tds__GetServices *tds__GetServices, _tds__GetServicesResponse &tds__GetServicesResponse)
{
    DEBUG_MSG("Device: %s\n", __FUNCTION__);

    ServiceContext* ctx = (ServiceContext*)this->soap->user;

    std::string XAddr = ctx->getXAddr(this->soap)

    //Device Service
    tds__GetServicesResponse.Service.push_back(soap_new_tds__Service(this->soap));
    tds__GetServicesResponse.Service.back()->Namespace  = "http://www.onvif.org/ver10/device/wsdl";
    tds__GetServicesResponse.Service.back()->XAddr      = XAddr;
    tds__GetServicesResponse.Service.back()->Version    = soap_new_req_tt__OnvifVersion(this->soap, 2, 5);
    if( tds__GetServices->IncludeCapability )
    {
        tds__GetServicesResponse.Service.back()->Capabilities        = soap_new__tds__Service_Capabilities(this->soap);
        tds__DeviceServiceCapabilities *capabilities                 = ctx->getDeviceServiceCapabilities(this->soap);
        tds__GetServicesResponse.Service.back()->Capabilities->__any = soap_dom_element(this->soap, NULL, "tds:Capabilities", capabilities, capabilities->soap_type());
    }

    tds__GetServicesResponse.Service.push_back(soap_new_tds__Service(this->soap));
    tds__GetServicesResponse.Service.back()->Namespace  = "http://www.onvif.org/ver10/media/wsdl";
    tds__GetServicesResponse.Service.back()->XAddr      = XAddr;
    tds__GetServicesResponse.Service.back()->Version    = soap_new_req_tt__OnvifVersion(this->soap, 2, 6);
    if (tds__GetServices->IncludeCapability)
    {
        tds__GetServicesResponse.Service.back()->Capabilities        = soap_new__tds__Service_Capabilities(this->soap);
        trt__Capabilities *capabilities                              = ctx->getMediaServiceCapabilities(this->soap);
        tds__GetServicesResponse.Service.back()->Capabilities->__any = soap_dom_element(this->soap, NULL, "trt:Capabilities", capabilities, capabilities->soap_type());
    }

    return SOAP_OK;
}

KoynovStas, 2019-06-25:
How do you start a daemon ?, give full start parameters. Also, to understand the problem, give the configuration of the network and network interfaces.

santhku, 2019-06-25:
Hi, 
started the daemon using below command
sudo ./onvif_srvd_debug  --ifs eth0 --scope onvif://www.onvif.org/name/santhosh --scope onvif://www.onvif.org/Profile/S --port 4000 --name RTSP --width 800 --height 600 --url rtsp://192.168.36.75:554/unicast --type H264 

configuration of the network and network interfaces.

eth0      Link encap:Ethernet  HWaddr 00:04:9f:05:fb:d4  
          inet addr:192.168.36.75  Bcast:192.168.36.255  Mask:255.255.255.0
          inet6 addr: fe80::204:9fff:fe05:fbd4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST DYNAMIC  MTU:1500  Metric:1
          RX packets:4239 errors:0 dropped:0 overruns:0 frame:0
          TX packets:43846 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:424873 (414.9 KiB)  TX bytes:105791479 (100.8 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:10459 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10459 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3217374 (3.0 MiB)  TX bytes:3217374 (3.0 MiB)

root# ip link show                              
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 00:04:9f:05:fb:d4 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 3000
    link/ether cc:c0:79:b4:85:df brd ff:ff:ff:ff:ff:ff
4: p2p0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 3000
    link/ether ce:c0:79:38:85:df brd ff:ff:ff:ff:ff:ff

root# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:04:9f:05:fb:d4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.36.75/24 brd 192.168.36.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::204:9fff:fe05:fbd4/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 3000
    link/ether cc:c0:79:b4:85:df brd ff:ff:ff:ff:ff:ff
4: p2p0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 3000
    link/ether ce:c0:79:38:85:df brd ff:ff:ff:ff:ff:ff

root# ip r 
default via 192.168.36.1 dev eth0 
4.2.2.2 via 192.168.36.1 dev eth0 
8.8.8.8 via 192.168.36.1 dev eth0 
172.16.137.30 via 192.168.36.1 dev eth0 
172.16.137.31 via 192.168.36.1 dev eth0 
192.168.36.0/24 dev eth0 proto kernel scope link src 192.168.36.75 
192.168.36.1 dev eth0 scope link 

Let me know how can i resolve this issue

KoynovStas, 2019-06-25:
Show me how `soap` determined the client's IP. output me `soap->ip`

santhku, 2019-06-28:
Im getting the soap->ip as below 

http://127.0.0.1:4000

KoynovStas, 2019-06-28:
You said that the port is not correct. but it is 4000, this is correct, because you pass it through command line parameters.

So soap itself could not determine your client's ip. which is why the `ServiceContext::getServerIpFromClientIp` function has returned the same address to you. or localhost as the default value! 
See how the method works: `ServiceContext::getXAddr`

santhku, 2019-07-03:
Thanks, 
I referred the getXAddr as per the definition, it should return the IP address of the server, 
In my case when i connect my hardware board to the network switch, i'm getting http://127.0.0.1:4000 when i using the getXAddr  function. 
But when i connect my hardware board directly to my PC ethernet port i am getting as  http://192.168.36.75:4000

please advice me any further test or changes i need to make

KoynovStas, 2019-07-06:
I do not see and do not know the configuration of your network. 
You need to understand how your network works. The `ServiceContext::getServerIpFromClientIp` function is not very complicated and I think you understand how it works. If you understand this, then you should understand that it tries to return the server IP address from the client IP address, IP is searched by mask on all network interfaces. 
Network switch, can not contain IP addresses! it was just a signal amplifier.

**Example:**
**PC**
 eth0: 192.168.1.1 mask 255.255.255.0
 eth1: 192.168.2.1 mask 255.255.255.0
 eth2: 10.30.1.1 mask 255.0.0.0
 
**Client:**
 eth0: 192.168.1.2 mask 255.255.255.0
 eth1: 192.168.2.2 mask 255.255.255.0
 eth2: 10.30.1.2 mask 255.0.0.0

**Configuration 1:**
PC eth0 <---> Client eht0;  
//soap->ip == 192.168.1.2
//getServerIpFromClientIp must be return 192.168.1.1

**Configuration 2:**
PC eth1 <---> Client eht1;  
//soap->ip == 10.30.1.2
//getServerIpFromClientIp must be return 10.30.1.1

**Configuration 3:**
PC eth2 <---> Client eht2;  
//soap->ip == 10.30.1.2
//getServerIpFromClientIp must be return 10.30.1.1

**Configuration 4:**
PC <---> Client on localhost;  
//soap->ip == 127.0.0.1
//getServerIpFromClientIp must be return 127.0.0.1

If you are using a NAT, router, or something else, contact your network administrator. Or try rewriting the `getServerIpFromClientIp` function to meet your requirements

santhku, 2019-07-11:
Thanks for detailed explanation, I will check as you have suggested