KoynovStas/onvif_srvd #11: How to add soapPTZBindingService

santhku, 2019-07-18:
Hi, 

I am trying to add the APPLY(PTZBindingService, soap) in onvif_srvd.c, i am getting the error as below

src/onvif_srvd.c:146:9: error: ‘PTZBindingService’ was not declared in this scope
   APPLY(PTZBindingService, soap)                   
         ^
src/onvif_srvd.c:171:40: note: in definition of macro ‘DECLARE_SERVICE’
 #define DECLARE_SERVICE(service, soap) service service ## _inst(soap);
                                        ^
src/onvif_srvd.c:424:5: note: in expansion of macro ‘FOREACH_SERVICE’
     FOREACH_SERVICE(DECLARE_SERVICE, soap)
     ^
src/onvif_srvd.c:146:9: error: ‘PTZBindingService_inst’ was not declared in this scope
   APPLY(PTZBindingService, soap)                   
         ^
src/onvif_srvd.c:174:26: note: in definition of macro ‘DISPATCH_SERVICE’
                 else if (service ## _inst.dispatch() != SOAP_NO_METHOD) {\
                          ^
src/onvif_srvd.c:441:9: note: in expansion of macro ‘FOREACH_SERVICE’
         FOREACH_SERVICE(DISPATCH_SERVICE, soap)
         ^
Makefile:160: recipe for target 'src/onvif_srvd_debug.o' failed
make: *** [src/onvif_srvd_debug.o] Error 1

How can i resolve this issue

KoynovStas, 2019-07-18:
You should read the comment in this file more carefully: [/src/onvif_srvd.c](./src/onvif_srvd.c)

santhku, 2019-07-18:
ok

santhku, 2019-07-19:
I found the below comment but, but getting errors while i do it.  Let me know how can i proceed ahead
/*
 * If you need support for other services,
 * add the desired option to the macro FOREACH_SERVICE.
 *
 * Note: Do not forget to add the gsoap binding class for the service,
 * and the implementation methods for it, like for DeviceBindingService

KoynovStas, 2019-07-21:
To add a new service, follow these steps:
1. You should learn the Makefile.
2. Read the [ONVIF specification](https://www.onvif.org/profiles/)
3. Add the WSDL file for the service: [https://www.onvif.org/profiles/specifications/](https://www.onvif.org/profiles/specifications/)
4. To generate a using gsoap files of the parser (the Makefile will do this automatically) you May need to clean
5. Add an implementation for the service (write handlers)
6. Add an implementation file to the Makefile.
7. Add the service to the macro FOREACH_SERVICE

You can read the early comments to understand how I've added say MediaBindingService.

santhku, 2019-07-22:
Thanks was able to add and compile,  now getting some errors i will try and resolve them

santhku, 2019-07-23:
Errors related to PTZ soap code and makefile changes I was able to do and I was able to create the PTZ binding services functions and now compile the code without any errors. 
But when i run the ONVIF client I am not able to get the PTZ functions enabled, 
I have created a PTZservice.cpp and added below code
int PTZBindingService::GetServiceCapabilities(_tptz__GetServiceCapabilities *tptz__GetServiceCapabilities, _tptz__GetServiceCapabilitiesResponse &tptz__GetServiceCapabilitiesResponse)
{
    DEBUG_MSG("PTZBindingService: %s\n", __FUNCTION__);

    ServiceContext* ctx = (ServiceContext*)this->soap->user;
    tptz__GetServiceCapabilitiesResponse.Capabilities = ctx->GetPTZServiceCapabilities(this->soap);
 
    return SOAP_OK;
}
and added the below code in ServiceContext.cpp

tptz__Capabilities *ServiceContext::GetPTZServiceCapabilities(soap *soap)
{
	tptz__Capabilities *capabilities = soap_new_tptz__Capabilities(soap);
	
	capabilities->EFlip       					= soap_new_ptr(soap, true);
	capabilities->Reverse       				= soap_new_ptr(soap, true);
	capabilities->GetCompatibleConfigurations   = soap_new_ptr(soap, true);
	capabilities->MoveStatus       				= soap_new_ptr(soap, true);
	capabilities->StatusPosition				= soap_new_ptr(soap, true);
	
}

but my PTZ button is still disabled, Do let me know i can enable ( I have attached screen shot)
[image]

santhku, 2019-07-31:
Hi, 
I was able to add the PTZ section and facing difficulty in the sending the response to ZoomSpeedSpace 
tptz__GetNodesResponse.PTZNode.back()->SupportedPTZSpaces->ZoomSpeedSpace.push_back(soap_new_tt__Space1DDescription(this->soap));

How can i send response  URI and XRange and YRange value (Min and Max) to the client

Tried the below, compiles fine but gives soap error when the code is run
struct tt__Space1DDescription* result = (struct tt__Space1DDescription*) my_soap_malloc(soap,sizeof(struct tt__Space1DDescription));
	result->URI=getContinuousZoomVelocitySpaceURI(soap);
	result->XRange = getFloatRange(soap, 10, 20);
	
	
	//std::string 
	tptz__GetNodesResponse.PTZNode.back()->SupportedPTZSpaces->ZoomSpeedSpace.push_back(result);

Let me know how can i proceed

KoynovStas, 2019-07-31:
I dont know. I did not work with PTZ. Try to examine the source of the gsoap parser.

santhku, 2019-08-01:
Thanks I reviewed gsoap parser and was able to solve

santhku, 2019-08-03:
I truly Appreciate your support so far PTZ  is now up and running 

Thanks ..

Now trying with PTZ presets

closed 2019-08-26 by santhku

roleoroleo, 2019-11-11:
Hi @santhku 
Could you share your work about PTZ?