KoynovStas/onvif_srvd #17: ONVIF website down.

stappersg, 2020-10-08:
Hi,

The README.md of this project has a link to  http://www.onvif.org  but that website is down.

This issue is now creating awareness of that.  So no need report here a broken link.

Checks like
* https://downforeveryoneorjustme.com/onvif.org?www=1
* https://isitdownorjust.me/onvif-org/
do confirm my observation of the site being off-line.

But https://down.com/  says the site is available.

Hopefully does reach this message the people  of http://www.onvif.org   ....

KoynovStas, 2020-10-09:
Everything works! Try using a proxy or VPN. Perhaps monkeys from your government are blocking sites about their corruption and attacks, and one of the addresses refers to the ONVIF site.

stappersg, 2020-10-09:
Yes, the site is now reachable.

closed 2020-10-09 by stappersg