KoynovStas/onvif_srvd #20: I need to start server using c#

Maruti-Lavate, 2020-10-21:
I want use this into c# project how i can start working on it. 
I want basicaly. When i call
OnvifServer.Start(); from c# should server start. So how i implement that OnvifServer in c# 
If any other reference plz mention 
Thank you in advance

KoynovStas, 2020-10-21:
This project is written in C++, do you want to use it in C#? Or do you want to run a binary? 
The service is implemented as a Linux daemon. just run the startup script.
Or do you want to implement such a service only using C#? Explain in more detail.

Maruti-Lavate, 2021-03-13:
Ok, thank you for interest 
C++ not supported in UWP its required c++/cx or WinRt, if i am right.
I have project which is run ip camera but its not supported onvif, that i want to give support with onvif to ip camera.
When any one searching camera in local network then my app respond soap message to other apps.
I think you getting my point what i want.