KoynovStas/onvif_srvd #26: Build error

yuzhimin999, 2020-10-24:
./SDK/gsoap.zip.tmp                                100%[===============================================================================================================>]  31.47M  10.6KB/s    surplus 0s    
Segmentation fault (core dumped)
make: *** [Makefile:216：generated/onvif.h] error 139

KoynovStas, 2020-10-24:
AND? Is this a description of the problem? Are you seriously?

This is an error in generating a file via gsoap. But explain to me how after downloading, we started generating the file, if we have to unpack it?

If so, then you have a failure to build the entire project. you have a dirty directory, you will have to manually delete the build files.

And so I will say that there is not enough data to help.
What did you do, how did you do it. what version was used.
You have problem with make or dependency problems or PC memory problem.

yuzhimin999, 2020-10-24:
I'm sorry, 
The version has not been updated to the latest version. I will try again when it is updated later.
You can close it, thanks a lot.

KoynovStas, 2020-10-24:
We will close this when you confirm that there are no build issues on the current (latest) version. And if there is, then this is a problem of the system environment and not Makefile.

yuzhimin999, 2020-10-30:
No build problem on the current (latest) version.
Thanks for you help

closed 2020-10-30 by KoynovStas