KoynovStas/onvif_srvd #27: The actual local IP address is 192.168.*.* ,but the program runs as follows, so the client cannot find it.

yuzhimin999, 2020-10-30:
The actual local IP address is 192.168.*.* ,but the program runs as follows, so the client cannot find it.

tcp        0      0 0.0.0.0:1000            0.0.0.0:*               LISTEN      10250/./onvif_srvd 
Please help to check the root cause,thanks a lot.

yuzhimin999, 2020-10-30:
> The actual local IP address is 192.168._._ ,but the program runs as follows, so the client cannot find it.
> 
> tcp 0 0 0.0.0.0:1000 0.0.0.0:* LISTEN 10250/./onvif_srvd
> Please help to check the root cause,thanks a lot.

yuzhimin999, 2020-10-30:
Failed to get IP address?

KoynovStas, 2020-10-30:
What is it: `tcp 0 0 0.0.0.0:1000 0.0.0.0:* LISTEN 10250/./onvif_srvd`?
Describe everything you do.
Give the network configuration, interface names and startup script parameters. As you will not understand, developers are not telepathic. We do not know how to read tarot cards and we do not have a crystal ball of predictions!

yuzhimin999, 2020-10-30:
by command:
$ netstat -ap
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
        
tcp        0      0 0.0.0.0:8090            0.0.0.0:*               LISTEN      3380/ffserver       
     
tcp        0      0 0.0.0.0:1000            0.0.0.0:*               LISTEN      3445/./onvif_srvd

yuzhimin999, 2020-10-30:
$ ifconfig 
enp0s3: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.41.203  netmask 255.255.255.0  broadcast 192.168.41.255

yuzhimin999, 2020-10-30:
./onvif_srvd  --ifs enp0s3 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP --width 800 --height 600 --url rtsp://0.0.0.0:8554/live1.h264 --type H264

yuzhimin999, 2020-10-30:
The Ubuntu18.4 system runs on the virtual machine VM
push stream by ffserver
ffserver -d -f /etc/ffserver.conf
ffmpeg -f v4l2 -i /dev/video0  -s 640x480 -r 24 -vcodec libx264 -an http://127.0.0.1:8090/feed1.ffm

yuzhimin999, 2020-10-30:
Can play "RTSP: / / 0.0.0.0:8554 / live1 h264" video stream by  VLC media player

yuzhimin999, 2020-10-30:
GSOAP_VERSION     = 2.8.93

KoynovStas, 2020-10-30:
use: --url rtsp://%s:8554/live1.h264!

yuzhimin999, 2020-10-30:
> use: --url rtsp://%s:8554/live1.h264!

by command
sudo ./onvif_srvd --ifs enp0s3 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP --width 800 --height 600 --url rtsp://%s:8554/live1.h264 --type H264

Still can not be found by the client, very confused@

KoynovStas, 2020-10-30:
Client use WS-Discovery? onvif_srvd does not contain WS-Discovery!

yuzhimin999, 2020-10-31:
> Client use WS-Discovery? onvif_srvd does not contain WS-Discovery!
Thank you for your support and patience reply.
I am using [ONVIF Device Manager ](https://sourceforge.net/projects/onvifdm/)of windows.

KoynovStas, 2020-10-31:
This application requires support for WS-Discovery.
You must either add this functionality to this template, or use the ready-made WS-Discover service. Alternatively, you can take mine:  https://github.com/KoynovStas/wsdd

Your problem is that you do not fully understand how it works. check out the onwif documentation.

yuzhimin999, 2020-11-01:
> This application requires support for WS-Discovery.
> You must either add this functionality to this template, or use the ready-made WS-Discover service. Alternatively, you can take mine: [...]

Thank you very much, your suggestion is very good, it is already working normally.

closed 2020-11-01 by yuzhimin999