KoynovStas/onvif_srvd #30: compile error

p3im4n, 2020-11-22:
hi
when I compile on ubunto 20.04,then I got this error:

make release

[build]  generated/soapC.o:
g++ -c generated/soapC.cpp -o generated/soapC.o  -s  -DDAEMON_NAME='"onvif_srvd"' -DDAEMON_MAJOR_VERSION=1 -DDAEMON_MINOR_VERSION=1 -DDAEMON_PATCH_VERSION=0 -DDAEMON_PID_FILE_NAME='"onvif_srvd.pid"' -DDAEMON_LOG_FILE_NAME='"onvif_srvd.log"' -DDAEMON_NO_CHDIR=1 -DDAEMON_NO_CLOSE_STDIO=0 -I./src -I./generated -I./gsoap-2.8/gsoap -I./gsoap-2.8/gsoap/custom -I./gsoap-2.8/gsoap/plugin -I./gsoap-2.8/gsoap/import -std=c++11 -O2  -Wall  -pipe
^[[B^[[B^[[A^[[B^[[C^[[B^[[D{standard input}: Assembler messages:
{standard input}:954873: Warning: end of file not at end of a line; newline inserted
{standard input}:955331: Error: number of operands mismatch for `sub'
{standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make: * [Makefile:156: generated/soapC.o] Error 1

any help would be great.

KoynovStas, 2020-11-23:
1.I never had this error
2. Perhaps you have a broken build sequence due to the settings (features) of the file system
3. Perhaps you should remove the `-pipe` parameter for the compiler, this will slow down the build, but force the compiler to save the temp files to disk.
4. You can contact the support for the gsoap, because this is the file that you received through the gsoap generator. (this is not my file).

p3im4n, 2020-11-23:
again, thx for your response Mr KoynovStas :)

p3im4n, 2020-11-23:
hi Mr @KoynovStas 
these are question when I want to open a ticket about my error in www.genivia.com:

wsdl2h command options used:
soapcpp2 command options used:
C/C++ compiler options used:

would you help me to answer???

best regards

KoynovStas, 2020-11-23:
you should examine the Makefile. all assembly steps and parameters are indicated there!

p3im4n, 2020-11-24:
hi,this is support's answer:
=================================================
Looks like the assembler fails:

 Assembler messages:
{standard input}:954873: Warning: end of file not at end of a line; newline inserted
{standard input}:955331: Error: number of operands mismatch for `sub'

This is likely a compiler bug or the compiler is not properly installed.
==================================================

so,would you tell me how install compiler properly?
sorry I'm windows user and have low experiences with linux and finally i have to port your great project to windows

best regards

KoynovStas, 2020-11-24:
I am using Kubuntu 20.04. Which is based on Ubuntu 20.04, only uses a different DM (KDE).
I didn't install the gcc compiler. It is installed on the system by default!
I assumed you are running low on RAM, and I advised you to remove the `-pipe` flag.
Or you are using a filesystem that make does not understand. which leads to the fact that the make begins to build, while the file is not fully generated yet!
I am using the classic EXT4 file system.
For work, I just installed additional libraries, it is written about this in the readme.

p3im4n, 2020-11-24:
hi Mr @KoynovStas
its so wired but i had to install gcc this way...
sudo apt install build-essential

best regard

p3im4n, 2020-11-24:
hi again Mr @KoynovStas
I compiled it successfully
and run this command:

~/onvif_srvd-master$ onvif_srvd  --ifs eth0 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP --width 800 --height 600 --url rtsp://%s:554/unicast --type JPEG"
> 

but the onvif camera is not discovered buy teseting application

whats the problem?

p3im4n, 2020-11-25:
hi Mr @KoynovStas
how can I debug this project,which IDE do you use?
would you give me the project files too?

best regards

KoynovStas, 2020-11-25:
This request was about a compilation error! And I thought you found a solution to this problem.
If so, why was this request not closed?

If you have a new problem, you need to open a new request.
It seems logical!

closed 2020-11-25 by p3im4n