KoynovStas/onvif_srvd #31: how to debug? which IDE?

p3im4n, 2020-11-25:
hi Mr @KoynovStas
how can I debug this project,which IDE do you use?
would you give me the project files too?

best regards

KoynovStas, 2020-11-25:
It seems to me that you are already starting to troll me.
If you ask such questions, then I will say that you do not need a debugger. You should be able to debug the system and isolate the error using the print function. This will turn your head on. If you use a debugger, you will fix the effect, not the cause!

I am using QtCreator.

p3im4n, 2020-11-25:
I dont know what  do you mean about troll you Mr @KoynovStas
I use debuging all the time on qt c++,c#.net,c++ .net,  and arm mcu!!

best regards

closed 2020-11-25 by KoynovStas