KoynovStas/onvif_srvd #32: the onvif_srvd is not detected by NVR

p3im4n, 2020-11-25:
hi again Mr @KoynovStas
I compiled it successfully
and run this command:

~/onvif_srvd-master$ onvif_srvd --ifs eth0 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTSP --width 800 --height 600 --url rtsp://%s:554/unicast --type JPEG"

but the onvif camera is not discovered buy testing application neither able to manually add by its ip ant its port!!
(by the way I connect by wifi not ethernet)
whats the problem?
 
best regard

KoynovStas, 2020-11-25:
Tell me how is your client trying to find our service? What protocol is used for this?

p3im4n, 2020-11-25:
I use "ONVIF DEVICE MANAGER" (your recommended test tool)
I press refresh button and nothing happens
as I know this tool must  search for http://192.168.1.105/onvif/device_service(192.168.1.105 is my system local ip)
another question is authentication,I thing onvif_svrd has none!

KoynovStas, 2020-11-25:
All (well, almost all) applications use WS-Discovery. This is written in the ONVIF documentation. i think you should read wikipedia. so that you have a complete understanding.

I will immediately answer your next question. this demon has no WS-Discovery.

p3im4n, 2020-11-25:
its has no WS_Discovery,so I must add it manually?

p3im4n, 2020-11-25:
> I will immediately answer your next question

it's your generousity Mr KoynovStas

KoynovStas, 2020-11-25:
I think you should look into the question, the documentation, and the source!
You are trying to find a service without WS-Discovery.
You can add it manually, but what do you want with it?

What will it give you? You are using wifi, but you have specified the interface as eth0.
What should I think about this? I think you are trying to figure it out, but for that you need to read the documentation.
Just from the outside, it looks like you don't understand what you are doing!

p3im4n, 2020-11-25:
1.I  try with cable before last comment 
2.you said that this demon has no WS-Discovery.,so tell me how do you  use it ?add it manually? how?

KoynovStas, 2020-11-25:
Examine these queries: #12 and #28
+ https://www.genivia.com/doc/wsdd/html/wsdd_0.html

p3im4n, 2020-11-25:
I m confuse ...
you said "this demon has no WS-Discovery."
then give me a link about  WS-Discovery :|

KoynovStas, 2020-11-26:
You have not figured out the question. You have not examined the requests that other people have already asked. (I showed them to you, where it is described that WS-discovery, I took it to a separate project) There is also a request that describes why there is no WS-security. You start testing, but do not understand what you are testing. You want to find a service without WS-discovery support. You are using wifi, but you specify eth0. The next question will be why there is no RTSP broadcast. Therefore, I ask you once again to understand how ONVIF works. 

I cannot give you links to the documentation every time to explain the terminology and principles of work. This information is available in the documentation and on the Internet.

This request is a duplicate #12, I am closing it!

I can't be your personal Google. You must enable head.

**Do not use this project, find an ONVIF for windows, it will be easier and more efficient!**

closed 2020-11-26 by KoynovStas