KoynovStas/onvif_srvd #33: wsdl2h generation stage

p3im4n, 2020-12-06:
hi Mr @KoynovStas 

how did u combine .wsdl and .xsd file to generate corresponding .h file  with generated wsdl2h.exe(at previos  stage) ?
 
abviously  these wsdl generate those .h:
devicemgmt.wsdl   ---> "soapDeviceBindingService.h"
media.wsdl             --->  "soapMediaBindingService.h"
ptz.wsdl                  --->  "soapPTZBindingService.h"

I saw in make file:

# ---- gSOAP ----

$(GENERATED_DIR)/onvif.h:
	@$(build_gsoap)
	@mkdir -p $(GENERATED_DIR)
	$(WSDL2H) -d -t ./wsdl/typemap.dat  -o $@  $(WSDL_FILES)
	$(WSSE_IMPORT)
........
......
and in typemap.dat:

#	OASIS recommended prefixes

wsnt	= <http://docs.oasis-open.org/wsn/b-2>
wsntw	= <http://docs.oasis-open.org/wsn/bw-2>
wsrfbf	= <http://docs.oasis-open.org/wsrf/bf-2>
wsrfr	= <http://docs.oasis-open.org/wsrf/r-2>
wsrfrw  = <http://docs.oasis-open.org/wsrf/rw-2>
wstop	= <http://docs.oasis-open.org/wsn/t-1>

the available xsd file in your project is are mentioned  there!

but i didnt understand the relation between their

best regards

p3im4n, 2020-12-06:
ok.. i concluded the .xsd file which are needed,automatically downloaded based on typemap.dat  information!!!
am i right?

KoynovStas, 2020-12-07:
about typemap.dat see: https://www.genivia.com/doc/guide/html/index.html

If .xsd file is located locally, then the link should be in brackets **<>**

p3im4n, 2020-12-07:
and what about generated file...?
you make this 3 file with wds2l.exe:
"soapDeviceBindingService.h"
"soapMediaBindingService.h"
"soapPTZBindingService.h"
t

then feed them to soapcpp2.exe to generate cpp file!
every time soapcpp2.exe generte these 6 files:

Saving XXXXBinding.nsmap namespace mapp(XXXX is dependet on wsdl filename)
Saving soapClient.c client call stub fun......
Saving soapClientLib.c client stubs with..
Saving soapServer.c server request dispa......
Saving soapServerLib.c server request di....
Saving soapC.c serialization functions...............

I read this example: 
https://www.genivia.com/examples/onvif/index.html

they combine all wdsl files and generate them one time .
in your case there is 3 .h file!
I concluded you rename the soapServer.c every time and compile....
ok,please describe in a paragraph.....

best regards