KoynovStas/onvif_srvd #34: Getting Started

jamesroyer, 2021-02-01:
I downloaded this project but I was not able to get it working immediately. I eventually got it working and thought I'd share how I did to save other people time and effort. Perhaps these suggestions can be added to the README or code.

To get the code to compile, I had to modify the Makefile and append `chrono_time_point.cpp` to the `SOAP_SRC` variable:
```
SOAP_SRC = ... $(GSOAP_CUSTOM_DIR)/chrono_time_point.cpp
```
To be able to test with `Lingodigit ONVIF Device Tool` I had to specify my computers IP (e.g. 192.168.1.99) as well as the port number used by `onvif_srvd` (i.e. 1000) in the `Add` dialog:
```
URI: http://192.168.1.99:1000/onvif/device_service
```
When starting `Lingodigit ONVIF Device Tool` it pops up a dialog:
```
Please check wsdd is in /usr/local/bin/wsdd
```
I put the `wsdd` executable in `/usr/local/bin/` and I also tried `/usr/local/bin/wsdd/`. I still get the error. I don't know why it doesn't work or how to fix it. At first I thought this might be why I couldn't connect to `onvif_srvd` but it doesn't seem to be needed. Once I added the port number, it worked.

KoynovStas, 2021-02-04:
You can tell in more detail.
What version (commit) did you build.
On which Operating System (name - version) you did it, which compiler did you use.

On OS Kubuntu 20.04, when use gcc 9.3, no additional changes are required.

Did you change the sources?
What version of gsoap did you use?

mleeman, 2021-04-21:
FWIW, I am packaging this software for buster; as part of the change, I have replaced the statically linked gsoap with the system libraries. 

You can find the packaging stuff in my debian/televic/buster branch on my fork (the same for wsdd btw).

If there is interest, I can publish them on some public apt repo.