KoynovStas/onvif_srvd #35: Сборка отваливается на этапе скачивания gsoap

Sheridan, 2021-03-17:
Клонирую репозиторий, запускаю `make release`, получаю вот такой результат.
```pi-camera-3dp onvif_srvd (master) # make release
--2021-03-17 16:28:01--  https://sourceforge.net/projects/gsoap2/files/gsoap-2.8/gsoap_2.8.92.zip/download
Resolving sourceforge.net (sourceforge.net)... 216.105.38.13
Connecting to sourceforge.net (sourceforge.net)|216.105.38.13|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-03-17 16:28:03 ERROR 404: Not Found.

--2021-03-17 16:28:03--  https://sourceforge.net/projects/gsoap2/files/oldreleases/gsoap_2.8.92.zip/download
Resolving sourceforge.net (sourceforge.net)... 216.105.38.13
Connecting to sourceforge.net (sourceforge.net)|216.105.38.13|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-03-17 16:28:04 ERROR 404: Not Found.

--2021-03-17 16:28:04--  https://master.dl.sourceforge.net/project/gsoap2/oldreleases/gsoap_2.8.92.zip
Resolving master.dl.sourceforge.net (master.dl.sourceforge.net)... 216.105.38.12
Connecting to master.dl.sourceforge.net (master.dl.sourceforge.net)|216.105.38.12|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.sourceforge.net/project/gsoap2/oldreleases/gsoap_2.8.92.zip [following]
--2021-03-17 16:28:05--  https://downloads.sourceforge.net/project/gsoap2/oldreleases/gsoap_2.8.92.zip
Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 216.105.38.13
Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|216.105.38.13|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-03-17 16:28:06 ERROR 404: Not Found.

make: *** [Makefile:220: generated/onvif.h] Error 8```
На sf доступен только https://sourceforge.net/projects/gsoap2/files/gsoap-2.8/gsoap_2.8.111.zip/download

Sheridan, 2021-03-17:
Пробую собирать с 2.8.111 ...

DCCLX, 2021-09-13:
Самостоятельно скачайте архив с последней версией из https://sourceforge.net/projects/gsoap2/files/ и положите в директорию SDK с именем gsoap.zip

KoynovStas, 2023-08-26:
Thanks, but updating a version every time it's removed from SourceForge is bad practice IMHO. I found a repository where copies of the gSoap archives are saved: https://github.com/SrcBackup/gsoap. When I have time to test the new version, I will update it, (at least it makes sense for the problem described here: https://sourceforge.net/p/gsoap2/bugs/1269/), this problem was solved in version 2.8.99.

closed 2023-08-26 by KoynovStas