KoynovStas/onvif_srvd #40: Update gsoap download url

MufasaZhang, 2021-11-22:
Hey, the gsoap download on the https://sourceforge.net has already changed, Add this code to makefile could fix 404 not found problem, 

wget -O ./SDK/gsoap.zip.tmp "https://sourceforge.net/projects/gsoap2/files/gsoap_$(GSOAP_VERSION).zip/download"             || \

and then, choose the right gsoap version.

EnchantedGoldenApple, 2023-08-14:
I changed 262th line in Makefile to `wget -O ./SDK/gsoap.zip.tmp "https://sourceforge.net/projects/gsoap2/files/latest/download"   || \` and it works. I dont know c++ and you write this 2 years ago, but maybe someone else will need this

KoynovStas, 2023-08-26:
Thanks, but updating a version every time it's removed from SourceForge is bad practice IMHO. I found a repository where copies of the gSoap archives are saved: https://github.com/SrcBackup/gsoap. When I have time to test the new version, I will update it, (at least it makes sense for the problem described here: https://sourceforge.net/p/gsoap2/bugs/1269/), this problem was solved in version 2.8.99.

closed 2023-08-26 by KoynovStas