KoynovStas/onvif_srvd #45: Unable to discover RPI 4 in Onvif device manager

NolanMaverick, 2023-10-11:
Hello,
         Thanks for making this opensource,  I'm trying to utilize this template in RPI 4 model B, without any issues I compiled and and ran this as Daemon, but I did not seen any issues, but I didn't seen my RPI 4 device getting detected in Onvif Device Manager.
         
         Note that through out process I just changed IP address and WIFI interface Name.

What could be the issue?

Regards,
Nolan

KoynovStas, 2023-10-11:
You need additionally: [WSDD - ONVIF WS-Discovery server](https://github.com/KoynovStas/wsdd)

NolanMaverick, 2023-10-11:
now its owrking, but getting NO SIGNAL in ODM, how to solve this issue.

KoynovStas, 2023-10-11:
What signal? I can't read minds! How am I supposed to guess what you are doing, what you want to get? ODM says the device has been found? Have you received a link to your RTSP server? This means ONVIF has done its job. ONVIF does not include a video transmission server!

NolanMaverick, 2023-10-11:
Sorry for undetailed explanation, Presennt I'm unable to stream VLC player also through rtsp
This is the issue I'm facing in VLC

**Your input can't be opened:
VLC is unable to open the Connection failed:
VLC could not connect to "10.10.102.127:554".
Your input can't be opened:
VLC is unable to open the MRL 'rtsp://10.10.102.127:554/unicast'.**

KoynovStas, 2023-10-11:
What does RTSP and ONVIF have to do with it?
Let's step by step:
1. WSDD is a service (web service) to find something on the network. In our case, it is needed so that it transmits the URL of our ONVIF server.
2. ONVIF server must transfer the URL to the stream server (let's assume RTSP)
2.1 The ONVIF server can, through certain requests, transmit or receive additional information, say, about Different URLs. or about different formats. or set the desired format, etc. BUT transmit or receive this information from/to the RTSP server.
3. ONVIF is not an image transfer server! He is not an RTSP server. RTSP server is a separate server!
4. ONVIF is a configuration and communication server. IT DOES NOT stream video! This must be done by another server.

Resume: 
If you found your ONVIF server using WSDD. Then WSDD did its job. If your client or Video Recorder received a link to the RTSP broadcast through the ONVIF server, then the ONVIF server has done its job.
Now the RTSP server should work.
If you want the RTSP server to be configured through the ONVIF server, YOU must implement this functionality yourself!

NolanMaverick, 2023-10-11:
Got it Thanks for your time, can you please suggest  me some resources and documentations to implement Profile T ONVIF server from RPI.

KoynovStas, 2023-10-11:
I don't deal with this topic, I can't say anything.
Well, if only banal things, they say, start with the [official specification](https://www.onvif.org/profiles-add-ons-specifications/), and [tests](https://www.onvif.org/profiles/conformance/device-test-2/) for it and examples of [gSOAP](https://www.genivia.com/examples/onvif/index.html), if you use it.
If I answered the question, please close this issues.

closed 2023-10-11 by NolanMaverick

NolanMaverick, 2023-10-11:
Thanks I ran separate RTSP server and  placed correct link, now its working in ONVIF device manager.