KoynovStas/onvif_srvd #48: Cross Compile

junheepark0310, 2023-11-21:
Hello! First of all, thank you for sharing the project. While doing the project I wanted to cross-compile and make this project available on my embedded board(arm7l). However, I am having a hard time figuring this out. Do you know or is there a specific command for cross-compiling in this project?

junheepark0310, 2023-11-21:
Oh Also I am having a error while compiling in 
cmake -B build . -DUSE_SYSTEM_GSOAP=1
cmake --build build

If I input this command this error log shows up.
Critical error: #import: Cannot open file "custom/duration.h" for reading.
Hint: use option -I<path> (for example -Igsoap/import:gsoap/custom:.)
CMakeFiles/onvif_srvd.dir/build.make:73: recipe for target 'generated/soapC.cpp' failed
make[2]: *** [generated/soapC.cpp] Error 1
CMakeFiles/Makefile2:85: recipe for target 'CMakeFiles/onvif_srvd.dir/all' failed
make[1]: *** [CMakeFiles/onvif_srvd.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

KoynovStas, 2023-11-21:
on Kubuntu 22.04:
```
cmake -B build . -DUSE_SYSTEM_GSOAP=1
-- Generator is set to: Unix Makefiles
-- No CMAKE_BUILD_TYPE specified, force set to: Release
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Searching gSOAP
-- GSOAP_VERSION         = 2.8.117
-- GSOAP_INCLUDE_DIR     = /usr/include
-- GSOAP_CUSTOM_DIR      = /usr/share/gsoap/custom
-- GSOAP_IMPORT_DIR      = /usr/share/gsoap/import
-- GSOAP_PLUGIN_DIR      = /usr/share/gsoap/plugin
-- GSOAP_WS_DIR          = /usr/share/gsoap/WS
-- GSOAP_C_LIBRARY       = /usr/lib/x86_64-linux-gnu/libgsoap.so
-- GSOAP_CK_C_LIBRARY    = /usr/lib/x86_64-linux-gnu/libgsoapck.so
-- GSOAP_SSL_C_LIBRARY   = /usr/lib/x86_64-linux-gnu/libgsoapssl.so
-- GSOAP_CXX_LIBRARY     = /usr/lib/x86_64-linux-gnu/libgsoap++.so
-- GSOAP_CK_CXX_LIBRARY  = /usr/lib/x86_64-linux-gnu/libgsoapck++.so
-- GSOAP_SSL_CXX_LIBRARY = /usr/lib/x86_64-linux-gnu/libgsoapssl++.so
-- GSOAP_SOAPCPP2        = /bin/soapcpp2
-- GSOAP_WSDL2H          = /bin/wsdl2h
-- Found gSOAP: /usr/lib/x86_64-linux-gnu/libgsoap.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/onvif_srvd/build
```
next:
```
cmake --build build
[  0%] Built target generate_version
[  6%] Creating gSOAP binding file
Saving /tmp/onvif_srvd/build/generated/onvif.h

**  The gSOAP WSDL/WADL/XSD processor for C and C++, wsdl2h release 2.8.117
**  Copyright (C) 2000-2021 Robert van Engelen, Genivia Inc.
**  All Rights Reserved. This product is provided "as is", without any warranty.
**  The wsdl2h tool and its generated software are released under the GPL.
**  ----------------------------------------------------------------------------
**  A commercial use license is available from Genivia Inc., contact@genivia.com
**  ----------------------------------------------------------------------------

Reading type definitions from type map "/tmp/onvif_srvd/wsdl/typemap.dat"
Reading '/tmp/onvif_srvd/wsdl/b-2.xsd'...
Done reading '/tmp/onvif_srvd/wsdl/b-2.xsd'
Reading '/tmp/onvif_srvd/wsdl/bf-2.xsd'...
Done reading '/tmp/onvif_srvd/wsdl/bf-2.xsd'
Reading '/tmp/onvif_srvd/wsdl/common.xsd'...
Done reading '/tmp/onvif_srvd/wsdl/common.xsd'
Reading '/tmp/onvif_srvd/wsdl/devicemgmt.wsdl'...
Done reading '/tmp/onvif_srvd/wsdl/devicemgmt.wsdl'
Reading '/tmp/onvif_srvd/wsdl/media.wsdl'...
Done reading '/tmp/onvif_srvd/wsdl/media.wsdl'
Reading '/tmp/onvif_srvd/wsdl/onvif.xsd'...
  Reading schema '/tmp/onvif_srvd/wsdl/common.xsd'...
  Done reading '/tmp/onvif_srvd/wsdl/common.xsd'
Done reading '/tmp/onvif_srvd/wsdl/onvif.xsd'
Reading '/tmp/onvif_srvd/wsdl/ptz.wsdl'...
Done reading '/tmp/onvif_srvd/wsdl/ptz.wsdl'
Reading '/tmp/onvif_srvd/wsdl/t-1.xsd'...
Done reading '/tmp/onvif_srvd/wsdl/t-1.xsd'

Note: option -p auto-enabled to generate wrappers for built-in types derived from xsd__anyType to support polymorphism in XML by (de)serializing any derived type of xsd:anyType (or xsd:anySimpleType) as elements annotated by xsi:type attributes in XML, use option -P to suppress and disable this feature

To finalize code generation, execute:
> soapcpp2 /tmp/onvif_srvd/build/generated/onvif.h
[... 37 lines cut ...]
Saving /tmp/onvif_srvd/build/generated/soapC.cpp serialization functions

Compilation successful 

[ 20%] Building CXX object CMakeFiles/onvif_srvd.dir/src/daemon.c.o
[ 26%] Building CXX object CMakeFiles/onvif_srvd.dir/src/onvif_srvd.cpp.o
[ 33%] Building CXX object CMakeFiles/onvif_srvd.dir/src/eth_dev_param.cpp.o
[ 40%] Building CXX object CMakeFiles/onvif_srvd.dir/src/ServiceContext.cpp.o
[ 46%] Building CXX object CMakeFiles/onvif_srvd.dir/src/ServiceDevice.cpp.o
[ 53%] Building CXX object CMakeFiles/onvif_srvd.dir/src/ServiceMedia.cpp.o
[ 60%] Building CXX object CMakeFiles/onvif_srvd.dir/src/ServicePTZ.cpp.o
[ 66%] Building CXX object CMakeFiles/onvif_srvd.dir/generated/soapC.cpp.o
[ 73%] Building CXX object CMakeFiles/onvif_srvd.dir/generated/soapDeviceBindingService.cpp.o
[ 80%] Building CXX object CMakeFiles/onvif_srvd.dir/generated/soapMediaBindingService.cpp.o
[ 86%] Building CXX object CMakeFiles/onvif_srvd.dir/generated/soapPTZBindingService.cpp.o
[ 93%] Building CXX object CMakeFiles/onvif_srvd.dir/usr/share/gsoap/custom/duration.c.o
[100%] Linking CXX executable onvif_srvd
[100%] Built target onvif_srvd
```

No Error!

I don’t know what version of gsoap you have on your system. If it is old, or too new, read the changes on the GSOAP website. If the system version is not suitable, build your own version, which is the default (2.8.92).

PS. I don’t understand, if you need cross-compilation, why are you using the system gsoap?

junheepark0310, 2023-11-22:
Thank you so much for your response! I am using the Ubuntu 16.04 version because that is the only version compatible with my arm7l IP cam board. Sorry for the confusing statement. Onvif is new to me so I probably misunderstood the gsoap concept.  If it is okay for you may I have your insight? I am currently doing a project which is converting RTSP stream to onvif stream so it is more secure. After cross-compiling I am going to make a firmware and apply it to my board so I can use this onvif application. 
I already did some work but I am not sure that I am going in the right way. 
First I already installed
sudo apt install flex bison byacc make cmake m4
sudo apt install openssl libssl-dev zlib1g-dev libcrypto++9v5 (I used libcrypto++9v5 because my version did not have libcrypto++8)

Then I built:
sudo cmake -B build .
sudo cmake --build build

sudo cmake -B build . -DUSE_GSOAP_STATIC_LIB=1
sudo cmake --build build

sudo cmake -B build -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_PROCESSOR=arm -DCMAKE_C_COMPILER=armlinux-gnueabihf-gcc -DCMAKE_CXX_COMPILER=arm-linux-gnueabihf-g++
sudo cmake --build build

Is there more builds that I need to include? As you said after cross-compiling 
I couldn't build:
sudo cmake -B build . -DUSE_SYSTEM_GSOAP=1

This gave me this kind of error:
CMake Error at /usr/local/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find gSOAP (missing: GSOAP_C_LIBRARY GSOAP_SSL_C_LIBRARY
  GSOAP_CXX_LIBRARY GSOAP_SSL_CXX_LIBRARY)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindgSOAP.cmake:184 (find_package_handle_standard_args)
  CMakeLists.txt:60 (find_package)

-- Configuring incomplete, errors occurred!

Please help me with your great insight. Thank you very much and have a nice day.

KoynovStas, 2023-11-22:
1. Why are you trying to use the system GSOAP if you have cross-compilation? You must compile your GSOAP for your IP camera with your toolchain (compiler for your ARM7). If you are working on Ubuntu for ARM7, then clarify how to install the system GSOAP on your system, this is not a question for me, this is a question for your distribution!
2. Сmake told you that HE **could not find** the system gsoap. You see the lines: `Could NOT find gSOAP`. You must install the system gsoap yourself, find information on your version of ubuntu.
3. What is the ONVIF stream? The task of ONVIF is to send the URL to your RTSP stream through (the fucking SOAP protocol) and that’s it. Yes, it can provide additional features, such as PTZ, BUT it does not stream video. HE just exchanges the URL with the DVR.

junheepark0310, 2023-11-22:
Thank you so much for your insight I will work on this have a nice day.

closed 2023-11-29 by junheepark0310