KoynovStas/onvif_srvd #49: Build error on arm-64 board

TRYOKETHEPEN, 2024-03-11:
Thanks for this project.

**Problem describe:**
I build this project on arm-64 board locally, using the system gSOAP.
The print info shows it did find gSOAP paths, but at next step it throws an error that can't open file "custom/duration.h".

**Board info:** 
Hardware: RK3588, armv8.
OS: Debian 11, Linux 5.10.160.

**Operate steps:**
(1) apt install gsoap libgsoap-dev.
(2) download the master branch zip, then unzip it and copy to board.
(3) make.

**Build logs:**
```
root@lubancat:/home/onvif_srvd-master# cmake -B build . -DUSE_SYSTEM_GSOAP=1
-- Generator is set to: Unix Makefiles
-- No CMAKE_BUILD_TYPE specified, force set to: Release
-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Searching gSOAP
-- GSOAP_VERSION         = 2.8.104
-- GSOAP_INCLUDE_DIR     = /usr/include
-- GSOAP_CUSTOM_DIR      = /usr/share/gsoap/custom
-- GSOAP_IMPORT_DIR      = /usr/share/gsoap/import
-- GSOAP_PLUGIN_DIR      = /usr/share/gsoap/plugin
-- GSOAP_WS_DIR          = /usr/share/gsoap/WS
-- GSOAP_C_LIBRARY       = /usr/lib/aarch64-linux-gnu/libgsoap.so
-- GSOAP_CK_C_LIBRARY    = /usr/lib/aarch64-linux-gnu/libgsoapck.so
-- GSOAP_SSL_C_LIBRARY   = /usr/lib/aarch64-linux-gnu/libgsoapssl.so
-- GSOAP_CXX_LIBRARY     = /usr/lib/aarch64-linux-gnu/libgsoap++.so
-- GSOAP_CK_CXX_LIBRARY  = /usr/lib/aarch64-linux-gnu/libgsoapck++.so
-- GSOAP_SSL_CXX_LIBRARY = /usr/lib/aarch64-linux-gnu/libgsoapssl++.so
-- GSOAP_SOAPCPP2        = /bin/soapcpp2
-- GSOAP_WSDL2H          = /bin/wsdl2h
-- Found gSOAP: /usr/lib/aarch64-linux-gnu/libgsoap.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/onvif_srvd-master/build
root@lubancat:/home/onvif_srvd-master# cmake --build build
Scanning dependencies of target generate_version
[  0%] Built target generate_version
[  6%] Creating gSOAP binding file
Saving /home/onvif_srvd-master/build/generated/onvif.h

**  The gSOAP WSDL/WADL/XSD processor for C and C++, wsdl2h release 2.8.104
**  Copyright (C) 2000-2020 Robert van Engelen, Genivia Inc.
**  All Rights Reserved. This product is provided "as is", without any warranty.
**  The wsdl2h tool and its generated software are released under the GPL.
**  ----------------------------------------------------------------------------
**  A commercial use license is available from Genivia Inc., contact@genivia.com
**  ----------------------------------------------------------------------------

Reading type definitions from type map "/home/onvif_srvd-master/wsdl/typemap.dat"
Reading '/home/onvif_srvd-master/wsdl/b-2.xsd'...
Done reading '/home/onvif_srvd-master/wsdl/b-2.xsd'
Reading '/home/onvif_srvd-master/wsdl/bf-2.xsd'...
Done reading '/home/onvif_srvd-master/wsdl/bf-2.xsd'
Reading '/home/onvif_srvd-master/wsdl/common.xsd'...
Done reading '/home/onvif_srvd-master/wsdl/common.xsd'
Reading '/home/onvif_srvd-master/wsdl/devicemgmt.wsdl'...
Done reading '/home/onvif_srvd-master/wsdl/devicemgmt.wsdl'
Reading '/home/onvif_srvd-master/wsdl/media.wsdl'...
Done reading '/home/onvif_srvd-master/wsdl/media.wsdl'
[... 108 lines cut ...]
This program is free software; you can redistribute it and/or modify it under
the terms of the GNU General Public License as published by the Free Software
Foundation; either version 2 of the License, or (at your option) any later
version.

This program is distributed in the hope that it will be useful, but WITHOUT ANY
WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE. See the GNU General Public License for more details.

You should have received a copy of the GNU General Public License along with
this program; if not, write to the Free Software Foundation, Inc., 59 Temple
Place, Suite 330, Boston, MA 02111-1307 USA

Author contact information:
engelen@genivia.com / engelen@acm.org

This program is released under the GPL with the additional exemption that
compiling, linking, and/or using OpenSSL is allowed.
--------------------------------------------------------------------------------
A commercial use license is available from Genivia, Inc., contact@genivia.com
--------------------------------------------------------------------------------
*/

extern typedef long long xsd__duration; /* duration in ms (1/1000 sec) */
```

KoynovStas, 2024-03-11:
I haven't tested versions older than 92. You are using 104. See the changes between these versions. For example, you should add an additional import path: https://sourceforge.net/p/gsoap2/bugs/1291/
We found it, but in version 92, there was no need to add it.

TRYOKETHEPEN, 2024-03-11:
> I haven't tested versions more than 92. You are using 104. See the changes between these versions. For example, you should add an additional import path: https://sourceforge.net/p/gsoap2/bugs/1291/ We found it, but in version 92, there was no need to add it.

Thanks for reply.
According the bug info, I've resolved this problem by changing CMakeLists.txt from
```
add_custom_command(
    OUTPUT ${GENERATED_DIR}/soapC.cpp
    COMMAND ${GSOAP_SOAPCPP2} -c++11 -j -L -x -S -d ${GENERATED_DIR}
            -I${GSOAP_INCLUDE_DIR}:${GSOAP_IMPORT_DIR} ${GENERATED_DIR}/onvif.h
    DEPENDS ${GENERATED_DIR}/onvif.h
    COMMENT "Creating gSOAP stubs and glue code"
)
```
to
```
add_custom_command(
    OUTPUT ${GENERATED_DIR}/soapC.cpp
    COMMAND ${GSOAP_SOAPCPP2} -c++11 -j -L -x -S -d ${GENERATED_DIR}
            -I/usr/share/gsoap/import
            -I/usr/share/gsoap
            -I${GSOAP_INCLUDE_DIR}:${GSOAP_IMPORT_DIR} ${GENERATED_DIR}/onvif.h
    DEPENDS ${GENERATED_DIR}/onvif.h
    COMMENT "Creating gSOAP stubs and glue code"
)
```

KoynovStas, 2024-03-11:
For more portable code use vars like ${GSOAP_IMPORT_DIR}
cmake found it:
```
-- Searching gSOAP
-- GSOAP_VERSION         = 2.8.104
-- GSOAP_INCLUDE_DIR     = /usr/include
-- GSOAP_CUSTOM_DIR      = /usr/share/gsoap/custom
-- GSOAP_IMPORT_DIR      = /usr/share/gsoap/import
-- GSOAP_PLUGIN_DIR      = /usr/share/gsoap/plugin
-- GSOAP_WS_DIR          = /usr/share/gsoap/WS
```