KoynovStas/onvif_srvd #51: How to start the onvif_srvd service of PTZ

wu12364564611, 2024-09-25:
Hello, I would like to know if I want to receive gimbal joystick information, then what parameters need to be configured in my onvif_srvd startup command? Can you provide a gimbal startup example? Thank you very much.

wu12364564611, 2024-09-26:
The following error occurred when I connected to the service: 
**SOAP 1.2 fault SOAP-ENV: MustUnderstand[no subcode]
"The data in element 'wsse:Security' must be understood but cannot be processed"
Detail: [no detail]**
What is the problem? Thank you for your answer.

KoynovStas, 2024-09-26:
I have not used wsse:Security, you should study the necessary mechanisms yourself. Perhaps in the specification for wsse:Security from ONVIF, the exchange protocol is described, but below, perhaps, the packages depend on the application. Understand ONVIF, this is an exchange, it does not even transmit a picture, this is done by RTSP. You should open the bindings you need and study their types.

wu12364564611, 2024-09-26:
I have successfully started the service and connected the PTZ joystick to our service. I have a question. Does the service only support the following PTZ parameters? Here is a picture of my joystick. In addition to moving forward, backward, left, and right, the joystick also has clockwise and counterclockwise rotation, and a button in the middle. Can these be supported?

- --ptz: 启用 PTZ（云台控制）支持。
- --move_left [value]: 设置调用 PTZ 左转动作的过程。
- --move_right [value]: 设置调用 PTZ 右转动作的过程。
- --move_up [value]: 设置调用 PTZ 上抬动作的过程。
- --move_down [value]: 设置调用 PTZ 下压动作的过程。
- --move_stop [value]: 设置调用 PTZ 停止动作的过程。
- --move_preset [value]: 设置调用 PTZ 转到预设位置动作的过程。
[image]

KoynovStas, 2024-09-26:
Well, look at how move_preset works, you can probably assign them or something like that, as I understand it, these are sets of preset movements.