KoynovStas/onvif_srvd #53: Which version of WSDL is used by this ONVIF project?

LunarStore, 2024-11-20:
Could you please let us know the version number of the WSDL used in this project? Our company's recent project plans to use your project, and we would like to know the ONVIF WSDL version being utilized. @KoynovStas

LunarStore, 2024-11-22:
I would like to know more about the WSDL used in this project, including those for media, device manager, and PTZ control. What API documentation should we refer to? Is there a more recent version of the official WSDL? Will there be any differences between the WSDL used in this project and the latest version? Which version of the documentation should we refer to when developing this project? Could you provide a link?

LunarStore, 2024-11-22:
> Um, well, the version of specific files is specified in the WSDL files themselves. I took the files here: https://www.onvif.org/profiles/specifications/ Or what version are you asking about?

I would like to know more about the WSDL used in this project, including those for media, device manager, and PTZ control. What API documentation should we refer to? Is there a more recent version of the official WSDL? Will there be any differences between the WSDL used in this project and the latest version? Which version of the documentation should we refer to when developing this project? Could you provide a link?

LunarStore, 2024-11-22:
I have ported the project to our product and used the following command:
`
./onvif_srvd --ifs eth0 --scope onvif://www.onvif.org/name/TestDev --scope onvif://www.onvif.org/Profile/S --name RTMP --width 1280 --height 512 --url rtmp://192.168.1.100:1935/live/substream --type H264 --no_fork --no_close
`
I can find device information in ODM, but I cannot view the RTMP stream. What should I do?

KoynovStas, 2024-11-22:
Sorry, but I still don't understand you. I gave you a link where I got the WSDL files. This is the official ONVIF website. There are both the WSDL files themselves and the documentation (PDF specification for these files)
Example:
device.wsdl file: https://www.onvif.org/ver10/device/wsdl/devicemgmt.wsdl
document that describes the API for it: https://www.onvif.org/specs/core/ONVIF-Core-Specification.pdf
If you are interested in the history of changes, it is also on the ONVIF website in the section: https://www.onvif.org/profiles/specifications/specification-history/

If you are asking about gsoap, then the documentation and examples are on the website: https://www.genivia.com/dev.html
Documentation for the server (daemon) itself is its source code.

It seems to me that you do not understand what ONVIF is. It is a communication and interaction protocol. It does not include an RTSP server or anything else. It is only a communication protocol between the camera and the video server. It does not transfer the data itself (audio, video, etc.).
If you are asking about ONVIF WS-Discovery server, then I have allocated it to a separate daemon: https://github.com/KoynovStas/wsdd

At the same time, these applications are needed only for communication (IP camera and video recorder), they do not do any real useful work. After finding the camera, it is asked for the RTSP address and that's it. Then RTSP works on its own and has nothing to do with ONVIF. The RTSP server is a separate server!

That's why I don't understand where the RTMP stream should come from. This is not the task of ONVIF. After you have received the link to the stream, the task of ONVIF is finished!. The stream should be issued by the server of this stream!

Short answer: You have to write a stream server. If you want to use RTSP. You have to write an RTSP server.

LunarStore, 2024-11-22:
> Sorry, but I still don't understand you. I gave you a link where I got the WSDL files. This is the official ONVIF website. There are both the WSDL files themselves and the documentation (PDF [...]

Yes, I understand. Onvif is a new concept to me. I know that Onvif is responsible for retrieving the parameter information of the stream, including the URL, width, and height. However, my question is: Does the Onvif ODM client support real-time playback of RTMP streams after obtaining the RTMP URL? Our lower-level machine runs the RTMP stream server.

LunarStore, 2024-11-22:
> Sorry, but I still don't understand you. I gave you a link where I got the WSDL files. This is the official ONVIF website. There are both the WSDL files themselves and the documentation (PDF [...]

Yes, I understand. Onvif is a new concept to me. I know that Onvif is responsible for retrieving the parameter information of the stream, including the URL, width, and height. However, my question is: Does the Onvif ODM client support real-time playback of RTMP streams after obtaining the RTMP URL? Our IPC runs the RTMP stream server.

LunarStore, 2024-11-22:
> If you're asking about this app: https://sourceforge.net/projects/onvifdm/ I don't know, ask its authors.

Sure, thank you so much for your response!

closed 2024-11-22 by LunarStore