KoynovStas/onvif_srvd #56: Trying to use onvif_srvd with Unifi Protect and it 500s on authentication attempts from Protect.

jameswhite, 2025-01-06:
I originally thought this was a [Unifi issue](https://community.ui.com/questions/Advanced-Adoption-Doesnt-send-traffic/1367b0eb-e26f-458f-8f6e-946b0bfa724b?page=0) but the following partial tcpdump indicates there's something wrong with onvif_srvd. hdovideo (the unifi protect NVR) chat about the date and time and then when it tries to authenticate with rainbow (where onvif_srvd is running) the service returns a 500.  I tried to make the conversation as readable as possible here. am I missing a setting or something? I ran `onvif_srvd` with: `/usr/local/sbin/onvif_srvd --no_fork --ifs wlan0 --port 80 --user admin --password admin --scope onvif://www.onvif.org/name/Rainbow --scope onvif://www.onvif.org/type/NetworkVideoTransmitter --scope onvif://www.onvif.org/Profile/Streaming --scope onvif://www.onvif.org/location/HighDesertOasis --scope onvif://www.onvif.org/hardware/RaspberryPiB --name RTSP --width 960 --height 540 --url rtsp://10.247.13.216:8554/unicast --type H264`

```
14:51:16.323658 IP hdovideo.localdomain.45628 > rainbow.localdomain.http: Flags [P.], seq 1:497, ack 1, win 502, options [nop,nop,TS val 832726358 ecr 2079747556], length 496: HTTP: POST /onvif/device_service HTTP/1.1
E..$..@.@.~....T.....<.P^..l................1.eV{.m.

POST./onvif/device_service.HTTP/1.1..Content-Type:.application/soap+xml;charset=utf-8;action="http://www.onvif.org/ver10/device/wsdl/GetSystemDateAndTime"..Content-Length:.261..charset:.utf-8..Host:.10.247.13.216..Connection:.close....

<s:Envelope.xmlns:s="http://www.w3.org/2003/05/soap-envelope">
  <s:Body.xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance".xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <GetSystemDateAndTime.xmlns="http://www.onvif.org/ver10/device/wsdl"/>
  </s:Body>
</s:Envelope>

14:51:16.326424 IP rainbow.localdomain.http > hdovideo.localdomain.45628: Flags [.], seq 1:1449, ack 497, win 32332, options [nop,nop,TS val 2079747564 ecr 832726358], length 1448: HTTP: HTTP/1.1 200 OK
E...m.@.@.........T.P.<....^....~LL.......{.m.
1.eVHTTP/1.1.200.OK..Server:.gSOAP/2.8..X-Frame-Options:.SAMEORIGIN..Content-Type:.application/soap+xml;.charset=utf-8;.action="http://www.onvif.org/ver10/device/wsdl/GetSystemDateAndTime"..Content-Length:.1536..Connection:.close....

<?xml.version="1.0".encoding="UTF-8"?>.
  <SOAP-ENV:Envelope.xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope".xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding".xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance".xmlns:xsd="http://www.w3.org/2001/XMLSchema".xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex".xmlns:wsa5="http://www.w3.org/2005/08/addressing".xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2".xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2".xmlns:xmime="http://tempuri.org/xmime.xsd".xmlns:xop="http://www.w3.org/2004/08/xop/include".xmlns:tt="http://www.onvif.org/ver10/schema".xmlns:wstop="http://docs.oasis-open.org/wsn/t-1".xmlns:tds="http://www.onvif.org/ver10/device/wsdl".xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl".xmlns:trt="http://www.onvif.org/ver10/media/wsdl">
    <SOAP-ENV:Body>
      <tds:GetSystemDateAndTimeResponse>
        <tds:SystemDateAndTime>
        <tt:DateTimeType>Manual</tt:DateTimeType>
        <tt:DaylightSavings>false</tt:DaylightSavings>
        <tt:TimeZone><tt:TZ>+00:00</tt:TZ></tt:TimeZone>
        <tt:UTCDateTime>
          <tt:Time>
            <tt:Hour>22</tt:Hour>
            <tt:Minute>51</tt:Minute>
            <tt:Second>16</tt:Second>
          </tt:Time>
          <tt:Date>
            <tt:Year>2025</tt:Year>
            <tt:Month>1</tt:Month>
            <tt:Day>6</tt:Day>
          </tt:Date>
        </tt:UTCDateTime>
        <tt:LocalDateTime>
          <tt:Time>
            <tt:Hour>22</tt:Hour>
            <tt:Minute>51</tt:Minute>
            <tt:Second>16</tt:Second>
          </tt:Time>
          <tt:Date>
            <tt:Year>2025</tt:Year>
            <tt:Month>1</tt:Month>
            <tt:Day>6</tt:Day>
          </tt:Date>
          </tt:LocalDateTime>
        </tds:SystemDateAndTime>
      </tds:GetSystemDateAndTimeResponse>
    </SOAP-ENV:Body>
  </SOAP-ENV:Envelope>..

14:51:16.351693 IP hdovideo.localdomain.45630 > rainbow.localdomain.http: Flags [P.], seq 1:1230, ack 1, win 502, options [nop,nop,TS val 832726386 ecr 2079747587], length 1229: HTTP: POST /onvif/device_service HTTP/1.1
E.../b@.@..{...T.....>.P..u:.*..............1.er{.n.POST./onvif/device_service.HTTP/1.1..Content-Type:.application/soap+xml;charset=utf-8;action="http://www.onvif.org/ver10/device/wsdl/GetServices"..Content-Length:.1002..charset:.utf-8..Host:.10.247.13.216..Connection:.close....

<s:Envelope.xmlns:s="http://www.w3.org/2003/05/soap-envelope".xmlns:a="http://www.w3.org/2005/08/addressing">
  <s:Header>
    <Security.s:mustUnderstand="1".xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd">
      <UsernameToken>
        <Username>admin</Username>
        <Password.Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">PlCAWLgiC5mkaiHhf6ueIqRstLo=</Password>
        <Nonce.EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">DxFzS8/TVomvJzQVxal0fg==</Nonce>
        <Created.xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">2025-01-06T22:51:16.000Z</Created>
      </UsernameToken>
    </Security>
  </s:Header>
  <s:Body.xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance".xmlns:xsd="http://www.w3.org/2001/XMLSchema">
[... 75 lines cut ...]
    </Security>
  </s:Header>
  <s:Body.xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance".xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <GetDeviceInformation.xmlns="http://www.onvif.org/ver10/device/wsdl"/>
  </s:Body>
</s:Envelope>

14:51:16.395598 IP rainbow.localdomain.http > hdovideo.localdomain.45634: Flags [P.], seq 1:1323, ack 1191, win 32038, options [nop,nop,TS val 2079747633 ecr 832726428], length 1322: HTTP: HTTP/1.1 500 Internal Server Error
E..^.N@.@..1.......T.P.B..,.1.<...}&........{.n11.e.HTTP/1.1.500.Internal.Server.Error..Server:.gSOAP/2.8..X-Frame-Options:.SAMEORIGIN..Content-Type:.application/soap+xml;.charset=utf-8..Content-Length:.1144..Connection:.close....

<?xml.version="1.0".encoding="UTF-8"?>.
  <SOAP-ENV:Envelope.xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope".xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding".xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance".xmlns:xsd="http://www.w3.org/2001/XMLSchema".xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex".xmlns:wsa5="http://www.w3.org/2005/08/addressing".xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2".xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2".xmlns:xmime="http://tempuri.org/xmime.xsd".xmlns:xop="http://www.w3.org/2004/08/xop/include".xmlns:tt="http://www.onvif.org/ver10/schema".xmlns:wstop="http://docs.oasis-open.org/wsn/t-1".xmlns:tds="http://www.onvif.org/ver10/device/wsdl".xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl".xmlns:trt="http://www.onvif.org/ver10/media/wsdl">
    <SOAP-ENV:Body>
      <SOAP-ENV:Fault>
        <SOAP-ENV:Code>
          <SOAP-ENV:Value>SOAP-ENV:MustUnderstand</SOAP-ENV:Value>
        </SOAP-ENV:Code>
        <SOAP-ENV:Reason>
          <SOAP-ENV:Text.xml:lang="en">The.data.in.element.'Security'.must.be.understood.but.cannot.be.processed</SOAP-ENV:Text>
        </SOAP-ENV:Reason>
      </SOAP-ENV:Fault>
    </SOAP-ENV:Body>
  </SOAP-ENV:Envelope>..

```

KoynovStas, 2025-01-07:
Well, what I see is that you requested **GetSystemDateAndTime**
You received the response: **GetSystemDateAndTimeResponse**
You requested **GetServices**, but not a simple one, but with **WS-Security** support. (with the user and password specified)

The server responded that it does not understand you: 
```
SOAP-ENV:Fault
<SOAP-ENV:Reason>
<SOAP-ENV:Text.xml:lang="en">The.data.in.element.'Security'.must.be.understood.but.cannot.be.processed</SOAP-ENV:Text>
</SOAP-ENV:Reason>
```
Error: **The.data.in.element.'Security'.must.be.understood.but.cannot.be.processed**

Sorry, I have not worked with **WS-Security**, you should figure out how to work with it yourself, At a minimum, it needs to be enabled during the build, see the readme:
`cmake -B build . -DWSSE_ON=1`

You must implement the verification part YOURSELF (password validation, etc. via SSL or similar). 
You must add this functionality yourself!
Refer to the ONVIF specification.

The following issues will also help you:
- [WS-Security Isn't working [I did compiled with WSSE_ON=1] #28](https://github.com/KoynovStas/onvif_srvd/issues/28)
- [I tried to use this server,find has no login ,can use this server. why?? #1](https://github.com/KoynovStas/onvif_srvd/issues/1)
- [with WSSE_ON=1 compilation failed #41](https://github.com/KoynovStas/onvif_srvd/issues/41)

jameswhite, 2025-01-08:
It looks like all that was needed was compiling with `-DWSSE_ON=1` no code changes. Here's a raspberry pi B running v4l2rtspserver and onvif_srvd all locally, imported into Unifi protect with admin/admin. awesome!
[image]

@KoynovStas thank you so much for this!

closed 2025-01-08 by jameswhite